TcMenu/LiquidCrystalIO #2: Backlight control issues (active low backlights & dimming)

bperrybap, 2020-05-24:
There are couple of issues with the current backlight API and the underlying control code.
As defined by LCD API 1.0, setbacklight() is an optional function.
https://playground.arduino.cc/Code/LCDAPI/
However, it is specified to be setbacklight(val)  where val represents a 0-255 brightness level.
val does not represent a backlight on/off state, nor a pin output state.
While there are some other "LiquidCrystal" implementations out there that make either of these assumptions, it is incorrect.

LCD API 1.0 was trying to come up with an API that worked across multiple h/w implementations and capabilities, a similar goal to this library's goal of supporting an io abstraction layer to support multiple ways to communicate with the LCD.
i.e. if the LCD library is LCD API 1.0 compliant, then the mainline sketch code can use backlight control API functions across different LCD devices without having to know which specific device is being used.
LCD API 1.0 took its goal one step further in that it was also trying to put any/all unique initialization information up in the constructor so that the user also didn't have to deal with any of that in the main line code.

Where backlight control for this library starts to be an issue, is when additional capabilities of backlight control is needed/wanted by the user's sketch.

Dimming is obvious as the current code and API won't support dimming.
The next issue is backlight active level.
Not all backlights are turned on by a logic high signal and off by a logic low signal.
This is particularly true on PCF8574 based backpacks.
Some backpacks use active low control.
The way the current LiquidCrystalIO code is written, there is no way to properly control an active low backlight using the backlight() and noBacklight() given the assumptions made in the code.

newLiquidCrystal, and hd44780 are two libraries that have LCD API 1.0 compliant backlight control.
newLiquidCrystal originally had a setBacklightPin(pin) function and simple backlight control like this library, the setBacklightPin() was udpated to take a "polarity" or active level parameter, and finally the code was re-factored to support dimming and be LCD API 1.0 compliant.
In that code while setBacklightPin(pin, pol) still exists, it is deprecated and should no longer be used as that information should be provided in the constructor.

I'm not sure the best approach is for this library, but there should at least be a way to support LCD devices that have backlight control circuits that use active LOW  to turn on the backlight.

davetcc, 2020-05-25:
I'm all for supporting standard API, spending a lot of time in the Java and C# world, most things have standardized API's and it makes coding much easier. I will make sure the library supports as much of the API as possible from  https://playground.arduino.cc/Code/LCDAPI/ before 1.3.0 is released.

Dealing with PWM over a I2C especially with PCF8574 seems like a particularly inefficient thing to do, especially given that the device cannot run in high-speed mode. Although it wouldn't be hard to add a task to taskManager that cycled the backlight state every few milliseconds, it would be highly inefficient. However, I can see that it would be useful if attached directly to an Arduino Pin. Maybe only supported on device local pins? 

Contrast Pin on PWM, yep many of the examples I added showed how to do this, but it seems to have been integrated into the API now. I have the code for this already and will add it as an extra function. 

All the libraries I manage are going through a process of standardization, I'm trying to get them all workable on a wide range of Arduino and Mbed devices. If there was an LCD library that worked on mbed and had an overridable delay function that made it work better with task manager, I'd switch tcMenu to using that, and suggest people use that instead.

bperrybap, 2020-05-25:
The h/w on i2c backpacks only supports on/off backlight control.
There is no way to do PWM over i2c as i2c is way too slow particularly at 100Khz SCLK.
The LCD API accommodates this:
```
setBacklight(val)
Set the backlight off/on, or set Backlight brightness (0-255)
If the display only has the option to turn the backlight on and off: 0 = off, >0 = on 
```

> If there was an LCD library that worked on mbed and had an overridable delay function that made it work better with task manager

I'm not sure what this means.
What do you mean by "overridable delay function" ?
I'm the author of the hd44780 library. I'd be open to looking at adding new capabilities.
I've not ever looked at mbed to see what, if any, changes are required to get it working in that environment.
hd44780 does have the ability to configure the LCD instruction timing.
Currently, hd44780 currently does not ever call yield() since that isn't available in all the Arduino IDE versions and core platforms that it has to run on so, "as is", it likely wouldn't work for your environment.
On the positive side hd44780 is about 4-5x faster at getting bytes transferred over i2c to the LCD so the net load on the processor is quite a bit less.

davetcc, 2020-05-25:
> I've not ever looked at mbed to see what, if any, changes are required to get it working in that environment.

I think it's coming our way anyway, regardless of IoAbstraction and tcMenu building for mbed, the new Nano boards are mbed under the covers, by the sound of it, they are running mbed 5 with RTOS as I understand.

>  What do you mean by "overridable delay function"

I mean that when the library is in a wait state for the LCD hardware, there is a way to override what it does, in the case of task manager, for example, that would be` taskManager.yieldForMicros(...)`. It would probably be best to either make the method of the class overridable, or provide a "delay" callback; which indicates the minimum number of micros to wait. 

> On the positive side hd44780 is about 4-5x faster at getting bytes transferred over i2c

The LCDs have a really slow refresh, any more than 2 updates a second make the display unreadable, I've tried various refresh rates with tcMenu and found anything more than that didn't work. So the best thing is to try and do other tasks while the 270khz protocol is being observed. Most displays are quite happy if the delay is longer than the specified minimum, so for example task manager could run other tasks updating the network, checking for changes in input, and doing other required logic and tests at the same time. This allows tcMenu to look quite responsive even on an AVR with the remote control protocol active over ethernet or serial. There's only one limitation, use only one task for rendering. Either a repeating game loop style task (tcMenu does this) or a carefully managed single triggered task.

bperrybap, 2020-05-26:
Going off issue here but
With respect to the transfer rate, the back of that sentence is what I was focusing on

> so the net load on the processor is quite a bit less

i.e. the overall load on the processor will be less with hd44780 than with LiquidCrystalIO
For a single byte transfer to the LCD LiquidCrystalIO will spend a bit over 1200us down in the Wire code.
hd44780 will spend just under 500us down in Wire.
LiquidCrystalIO breaks up that time in roughly 200us intervals where it calls the task scheduler whereas hd44780 will run (block) the full 500us +49us of other overhead before returning.

The real issue is the Print class. The print class calls the i/o library character by character until it is done. So the caller is blocked the entire amount of time it takes to output **all** the characters.
What would help would be Print could call a scheduler, or if the i/o library would call some kind of scheduling function in its write() code.
In the case of hd44780, it could do something  just before it does what it does now.
That is a very simple tweak that could offer worst case 550us scheduling intervals at 100Khz clock and 199us intervals on a 400Khz i2c clock.

--------------------------------
A bit more about hd44780

hd44780 is very efficient how handles the LCD instruction timing and hd44780_I2Cexp is much more efficient how it utilizes the i2c bus to transfer bytes to the LCD.
It will only ever do a delay if it is necessary. i.e the code is smart enough to keep track of the elapsed time and know how long a h/w interface takes to transmit the bits to the LCD and so it will only do added delays when needed. It doesn't do what other LCD libraries do with the blind delay just after sending the instruction to the LCD.
The library returns immediately after sending a command or byte to the LCD. 
If the next byte or command comes in and a delay is needed it will wait but on i2c even at 400Khz a delay is only necessary on home() or clear() instructions if they come in before the execution time has elapsed.

hd44780_I2c minimizes i2c bus utilization when transferring bytes to the lcd.
LiqiudCrystalIO will to do two i2c messages each has 3 bytes to transfer a single byte to the LCD.
hd44780 will do a single i2c message write of 4 bytes to transfer a single byte to the LCD.

On a 16mhz AVR, with a 100kHz i2c clock, LiquidCrystalIO will transfer a single byte to the LCD in 1706us hd44780 does does the LCD byte transfer in 549us.
A single i2c byte transfer is around 209us. 6 transfers is 1.2ms waiting down in Wire.
hd44780 does the LCD byte transfer in a single 4 byte i2c transfer that is is 493us.
hd44780 spends 493us down in Wire vs LiquidCrystalIO will spend 1200us down in Wire.

The difference is that LiquidCrystalIO can reschedule at least every 200us including when there are multiple data bytes. With hd44780 the block is 500us per character written to the display in a single print()

davetcc, 2020-05-26:
I agree completely that hd44780 is a far better library that's been well-thought through, and to be honest, LiquidCrystalIO is a library I'd rather not have had to write. I only forked the library because LiquidCrystal devices are so slow to interface with that it **heavily** affected task manager performance, none of the other displays (including colour TFT's) really cause this effect because they use the respective bus at full speed.

I'll say strongly here that I only recommend the use of this library where task manager is already being heavily used, for example with tcMenu. and nothing would make me happier than to be able to decommission it.

I don't think mbed support is really needed to be honest, most of those boards are 3V3 and you'd have to jump through hoops with voltage converters to use one. I just tried and it's not fun.

I'd be happy to revert the changes back to a couple of days ago and look at options for using your library. The quickest way is for me not to use the print function in tcMenu, using write instead, I could even call my yield function in between each write operation.

bperrybap, 2020-05-26:
pm me on the Arduno forum. Perhaps we can work out something for hd44780 that might be of benefit for the task manager, particularly if you could hook into the Arduino yield() support.

bperrybap, 2020-05-26:
BTW, there are a couple of very small things you can do to LiquidCrystalIO (mostly deleting a few lines of code which is doing things that are unnecessary) in pulseEnable() code that can speed things up quite a bit.
i.e. no need to do the initial write of E to low. (it is always left in low state so setting it to low again is not necessary)
The delay in pulseEnable() should not be there. It  should be moved up to send() as it is really a delay for the completion of all LCD instructions other than clear() and home(). There is no delay needed between nibbles.
The delay only needs to be 37us for standard LCDs, so 40us is plenty.
(on i2c it could be completely eliminated given the transfer time of the bytes over the bus)

davetcc, 2020-06-30:
I'll try and continue to move towards the API spec that we discussed earlier, most of the things discussed above have been fixed in this release. I'll raise another ticket to try and support as much of the api as possible.

closed 2020-06-30 by davetcc