TcMenu/LiquidCrystalIO #3: EepromAbstractionWire.cpp fails to compile on chipKit pic32 core

bperrybap, 2020-05-24:
There is an issue compiling EepromAbstractionWire.cpp on the chipKITt pic32 core.
This prevents the library from being used on that core even if the EEPROM code is not being used.
It looks like the issue is a Wire library incompatibility in that the chipKIT core is lacking one of the requestFrom() API functions being used.
Given the code is always calling requestFrom(address, quantity, sendstop) with a sendstop of true:

`wireImpl->requestFrom(eepromAddr, (uint8_t)1, (uint8_t)true);`

` wireImpl->requestFrom(eepromAddr, (uint8_t)currentGo, (uint8_t)true);`

It could revert to using the two argument version of requestFrom(address, quantity) instead.
i.e. change those lines to:
`wireImpl->requestFrom(eepromAddr, (uint8_t)1);`

` wireImpl->requestFrom(eepromAddr, (uint8_t)currentGo);`

bperrybap, 2020-05-24:
Something else to consider is that including <Wire.h> will cause a good portion of the Wire library code to be pulled in regardless of whether it is ever used. This is because it uses interrupts and any code that uses interrupts will always be linked in as the linker is unable to do unused/dead code removal on code that uses ISRs.
This is a difficult thing to work around. Very similar to the issue of virtual functions always getting linked in even if not ever used.

davetcc, 2020-05-25:
Agreed, I have been mulling over the possibility of using the direct TWI functions in both the device abstraction and EEPROM abstraction instead. Maybe writing a very small wrapper class. 

The wrapper could then be eventually re-written on some common architectures to yield to task-manager while waiting. This is the long term plan for the mbed version too, where it's even more important, with a 180mhz STM32-F439 board, a few millis waiting for IO is a huge number of instructions.

For now, I'll raise an issue in IoAbstraction for the next release to at least make it compile.

davetcc, 2020-06-30:
Fixed in 1.5.0 of IoAbstraction

closed 2020-06-30 by davetcc