TcMenu/LiquidCrystalIO #9: Setbacklight does not sync ioDevice

kimble4, 2021-03-14:
Using an LCD via a PCF8574 on i2c, with the usual arrangement of the backlight being driven by pin 3.  If you call setBacklight(), the change does not take effect until the next LCD command.

This appears to be due to the use of:
`ioDeviceDigitalWrite(_io_method, _backlightPin, state);`

...rather than the synchronised method.  This is a problem if you simply want to timeout (or illuminate) the backlight without changing the contents of the display.  I'm issuing a setCursor() after setting the backlight status as a work-around.

davetcc, 2021-03-15:
Thanks for reporting, we'll get that fixed and released soon.

davetcc, 2021-05-11:
Sorry it took a bit longer than usual, released today.