TcMenu/LiquidCrystalIO #10: Use of analogueWrite() prevents compile on ESP32

kimble4, 2021-04-06:
This use of analogWrite() appears to be the only thing preventing the library from compiling and working on ESP32, albeit without PWM backlight.  Presumably this could be fixed in PlatformDetermination.h or something.

If you wanted PWM you'd need to use ledcWrite(), which is  a bit more involved.

```
void LiquidCrystal::setBacklight(uint8_t state) {
    if (_backlightPin == 0xff) return;

#ifdef IOA_USE_ARDUINO
    if (_backlightMode == BACKLIGHT_PWM) {
        analogWrite(_backlightPin, state);
        return;
    }
#endif
    if (_backlightMode == BACKLIGHT_INVERTED) state = !state;

    ioDeviceDigitalWrite(_io_method, _backlightPin, state);
}

```

davetcc, 2021-04-09:
Hi there, thanks for raising the issue. 

We already have an analog abstraction that works completely on mbed (using the PwmIn/PwmOut/AnalogOut class), ESP32 (using DAC and LEDC calls) and all other Arduino devices. It will be a trivial activity to move this class over to use that, I'll try and do it later today. For now, I guess you can just comment that line out.

In fact, I'll expose an additional method where the AnalogDevice* can be provided too, then in future when we support it, you could even put a cheap i2c potentiometer/dac on an I2C backpack and support it there.

Dlloydev, 2021-04-15:
You might find this helpful (I hope). I've just created a small analogWrite library that I've tested OK using an ESP32 dev board .... (awaiting for delivery of an ESP32-S2 board) so an update might be necessary. Works with 8 PWM and 2 DAC pins ... resources are automatically managed.

[image]      [documentation](https://github.com/Dlloydev/ESP32-ESP32S2-AnalogWrite)

davetcc, 2021-04-15:
Thanks for that, but we already have an analog abstraction for all supported boards including ESP32 and mbed in IoAbstraction, it supports DAC and PWM pins, but I just didn't get around to moving this library across to use it until now.

https://github.com/davetcc/IoAbstraction/blob/master/src/AnalogDeviceAbstraction.h