TcMenu/LiquidCrystalIO #11: Declaring LiquidCrystalI2C causes immediate crash on ESP32

kimble4, 2022-06-23:
I've been happily using this library with version 1.0.6 of the ESP32.  I recently upgraded to ESP32 V2.0.3, and now find that the device crashes immediately:

```
ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0

assert failed: xQueueGenericSend queue.c:820 (pxQueue)

Backtrace:0x40083515:0x3ffe39500x400881a5:0x3ffe3970 0x4008d0fd:0x3ffe3990 0x4008899a:0x3ffe3ac0 0x400d26ae:0x3ffe3b00 0x400d193d:0x3ffe3b20 0x400d16e5:0x3ffe3b50 0x400d12c1:0x3ffe3b70 0x400d1329:0x3ffe3b90 0x400d138c:0x3ffe3bb0 0x400d125a:0x3ffe3be0 0x400da64f:0x3ffe3c10 0x40082dc5:0x3ffe3c40 0x4007920a:0x3ffe3c90  |<-CORRUPTED
```

Stripping things down to basics, this appears to happen simply as a result of the

`LiquidCrystalI2C_RS_EN(lcd, 0x20, false)`

declaration; it never gets as far as lcd.begin().

<IoAbstractionWire.h> seems okay; I'm using this to control another PCF8574 without problems.

davetcc, 2022-06-23:
Thanks for letting me know, I'll try and build with this version soon and see how it goes.

davetcc, 2022-07-13:
I think I found the problem, apologies been a bit hectic recently

davetcc, 2022-07-13:
Given the nature of the issue im going to release immediately please close if fixed

closed 2022-10-03 by davetcc