TcMenu/LiquidCrystalIO #14: cannot use two different LCD with PCF8574 in the same arduino sketch

albmef, 2022-11-19:
Hi
if i try to initialize two LCD with differents addresses on the same bus i get a compilation error
"
LiquidCrystalI2C_RS_EN(lcd0, 0x27, false)
LiquidCrystalI2C_RS_EN(lcd1, 0x26, false)
"

looks like something in macro expansion went wrong because "macro_backlight_tc" seems to have been defined twice imho

"
In file included from /tmp/.arduinoIDE-unsaved20221019-3301-1y9wb1.lcm1wi/HelloI2c/HelloI2c.ino:27:
/home/alban/Arduino/libraries/LiquidCrystalIO/src/LiquidCrystalIO.h:397:10: error: conflicting declaration 'auto macro_backlight_tc'
  397 |     auto macro_backlight_tc = (backlightInv) ? LiquidCrystal::BACKLIGHT_INVERTED : LiquidCrystal::BACKLIGHT_NORMAL; \
      |          ^~~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved20221019-3301-1y9wb1.lcm1wi/HelloI2c/HelloI2c.ino:40:1: note: in expansion of macro 'LiquidCrystalI2C_RS_EN'
   40 | LiquidCrystalI2C_RS_EN(lcd1, 0x26, false)
      | ^~~~~~~~~~~~~~~~~~~~~~
/home/alban/Arduino/libraries/LiquidCrystalIO/src/LiquidCrystalIO.h:397:10: note: previous declaration as 'LiquidCrystal::BackLightPinMode macro_backlight_tc'
  397 |     auto macro_backlight_tc = (backlightInv) ? LiquidCrystal::BACKLIGHT_INVERTED : LiquidCrystal::BACKLIGHT_NORMAL; \
      |          ^~~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved20221019-3301-1y9wb1.lcm1wi/HelloI2c/HelloI2c.ino:39:1: note: in expansion of macro 'LiquidCrystalI2C_RS_EN'
   39 | LiquidCrystalI2C_RS_EN(lcd0, 0x27, false)
      | ^~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
"

arduino IDE v2.0.2
raspberrypi pico board
LiquidCrystalIO v 1.4.3

best regards,
Alban

davetcc, 2022-11-21:
Just create them manually instead if you need more than one, without using the macro. If you `#include <IoAbstractionWire.h` then you only need:

```
    auto macro_backlight_tc = (backlightInv) ? LiquidCrystal::BACKLIGHT_INVERTED : LiquidCrystal::BACKLIGHT_NORMAL;
    LiquidCrystal var(0, 1, 2, 4, 5, 6, 7, macro_backlight_tc, ioFrom8574(addr, 0xff, &Wire, false);
```

The macros is simply a convenience for the easy cases.

enord, 2023-04-20:
Hi, I have the same issue, needing to display on two displays simultaneously, however I don't understand what I need to do...
I am using #include <IoAbstractionWire.h>
Where do I put your answer?
How do I address the different displays?
I had previously tried putting the following two instances:
     LiquidCrystalI2C_RS_EN(DisplayLCD1, 0x22, false)   and
    LiquidCrystalI2C_RS_EN(DisplayLCD2, 0x23, false) 
using:
    DisplayLCD1.begin(20, 4);
    DisplayLCD2.begin(20, 4);
    
     // most backpacks have the backlight on pin 3.
    DisplayLCD1.configureBacklightPin(3);
    DisplayLCD2.configureBacklightPin(3);
    DisplayLCD1.backlight();
    DisplayLCD2.backlight();   
 
 and referring to either DisplayLCD1 or DisplayLCD2:  eg. DisplayLCD1.print("hello over i2c!");
 
 I suspect I need to do something with the:     
   DisplayLCD2.configureBacklightPin(3);
   DisplayLCD1.backlight();
 
 but I cannot work out what it is I need to change!   Any help very much appreciated!

davetcc, 2023-04-20:
You literally declare the variable yourself instead of using the macro. The code just showed what the macro actually does.